boltsparts/BOLTS_archive #148: import BOLTS (yosemite) fails on yaml

prussiap, 2015-01-13:
Hi,
I've been reading through the forums, github repo etc and finally got Bolts to import from the python console on OSX yosemite FreeCad..

I installed yaml in python, using the same version that freecad uses.
Freecad Python 2.7.5 
OSX: Python 2.7.6

pyyaml, libyaml is installed in console :
Requirement already up-to-date: PyYAML in /Library/Python/2.7/site-packages/PyYAML-3.11-py2.7-macosx-10.10-intel.egg

It seems i can import yaml from python console but not freeCAD. Everything seems to be point to the proper bin for python etc.. 

any thoughts?
Traceback (most recent call last):
  File "<input>", line 1, in <module>
ImportError: No module named yaml

jreinhardt, 2015-01-14:
Thank you for reporting this. I have no access to an OS X system, so I would be very grateful if we could work this out together.

Can you type

    import sys
    for p in sys.path:
         print p

into the python console. This should print a list of all paths where FreeCAD looks for python stuff. On Linux, this includes the standard locations for python packages.

prussiap, 2015-01-14:
Hi,
With pleasure.. I'm a software dev so i'd like to find out.. Weird so are those all symlinks to system python because it's weird that FreeCad would have python in it's Contents/lib folder (sorry censored user folder):

/Applications/FreeCAD.app/Contents/Mod
/Applications/FreeCAD.app/Contents/lib
/Applications/FreeCAD.app/Contents/Mod/Drawing
/Applications/FreeCAD.app/Contents/Mod/Inspection
/Applications/FreeCAD.app/Contents/Mod/Idf
/Users/pru**/Library/Preferences/FreeCAD/Mod/BOLTS2
/Applications/FreeCAD.app/Contents/Mod/Part
/Applications/FreeCAD.app/Contents/Mod/Robot
/Users/pru**/Library/Preferences/FreeCAD/Mod/start_bolts.FCMacro
/Applications/FreeCAD.app/Contents/Mod/Arch
/Users/pru**/Library/Preferences/FreeCAD/Mod/.DS_Store
/Applications/FreeCAD.app/Contents/Mod/Mesh
/Applications/FreeCAD.app/Contents/Mod/Raytracing
/Applications/FreeCAD.app/Contents/Mod/Material
/Applications/FreeCAD.app/Contents/Mod/Web
/Applications/FreeCAD.app/Contents/Mod/Complete
/Applications/FreeCAD.app/Contents/Mod/Import
/Applications/FreeCAD.app/Contents/Mod/Test
/Applications/FreeCAD.app/Contents/Mod/Draft
/Applications/FreeCAD.app/Contents/Mod/Points
/Applications/FreeCAD.app/Contents/Mod/Start
/Applications/FreeCAD.app/Contents/Mod/Spreadsheet
/Applications/FreeCAD.app/Contents/Mod/Fem
/Applications/FreeCAD.app/Contents/Mod/OpenSCAD
/Applications/FreeCAD.app/Contents/Mod/MeshPart
/Applications/FreeCAD.app/Contents/Mod/Plot
/Users/pru**/Library/Preferences/FreeCAD/Mod/BOLTS
/Applications/FreeCAD.app/Contents/Mod/Ship
/Applications/FreeCAD.app/Contents/Mod/PartDesign
/Applications/FreeCAD.app/Contents/Mod/Image
/Applications/FreeCAD.app/Contents/Mod/ReverseEngineering
/Applications/FreeCAD.app/Contents/Mod/Sketcher
/Applications/FreeCAD.app/Contents/bin
/Applications/FreeCAD.app/Contents/lib/python2.7/site-packages/setuptools-0.9.7-py2.7.egg
/Applications/FreeCAD.app/Contents/lib/python2.7/site-packages/pip-1.4-py2.7.egg
/Applications/FreeCAD.app/Contents/lib/python2.7/site-packages/pyparsing-2.0.1-py2.7.egg
/Applications/FreeCAD.app/Contents/lib/python2.7/site-packages/tornado-3.2-py2.7-macosx-10.6-x86_64.egg
/Applications/FreeCAD.app/Contents/lib/python2.7/site-packages/python_dateutil-2.2-py2.7.egg
/Applications/FreeCAD.app/Contents/lib/python2.7/site-packages/backports.ssl_match_hostname-3.4.0.2-py2.7.egg
/Applications/FreeCAD.app/Contents/lib/python2.7/site-packages/six-1.5.2-py2.7.egg
/Applications/FreeCAD.app/Contents/lib/python2.7/site-packages/matplotlib-1.3.1-py2.7-macosx-10.6-x86_64.egg
/
/Applications/FreeCAD.app/Contents/lib/python27.zip
/Applications/FreeCAD.app/Contents/lib/python2.7
/Applications/FreeCAD.app/Contents/lib/python2.7/plat-darwin
/Applications/FreeCAD.app/Contents/lib/python2.7/plat-mac
/Applications/FreeCAD.app/Contents/lib/python2.7/plat-mac/lib-scriptpackages
/Applications/FreeCAD.app/Contents/lib/python2.7/lib-tk
/Applications/FreeCAD.app/Contents/lib/python2.7/lib-old
/Applications/FreeCAD.app/Contents/lib/python2.7/lib-dynload
/Applications/FreeCAD.app/Contents/lib/python2.7/site-packages
/Users/pru**/Library/Preferences/FreeCAD/Mod
/Users/pru**/Library/Application Support/FreeCAD/Mod

jreinhardt, 2015-01-14:
So /Applications/FreeCAD.app/Contents/lib/python2.7/site-packages is a symlink to "/Library/Python/2.7/site-packages/"?

Then I believe it should work, shouldn't it? I will ask the MacOS specialist of FreeCAD if there is anything special.

prussiap, 2015-01-14:
Yeah it was a symlink as far as I can tell. it definitly should work. if I just go into the python console i can import libyaml, I also see the library installed. 

Also the Bolts script was a bit different to install for me. I had to try a few incarnations of what others did on the forum. Not sure if that's because of Yosemite though.. 

I'd love to help get this working. I've been dying to try it out to see if i can make some helping hand for soldering with FreeCad :)

Feel free to email me also if you want to debug faster.

jreinhardt, 2015-01-14:
Ok, so in FreeCADs python console you can do

    import yaml

and it doesn't throw an import error? Or in a ordinary python shell?

Which version of BOLTS do you use?

prussiap, 2015-01-14:
no it fails from FreeCAD python console but works in the python shell.

jreinhardt, 2015-01-14:
I see, so it seems that for some reason it doesn't pick up the symlinks. Could you try, if it works when you add the directory directly to the path, i.e. in FreeCAD python console

    import sys
    sys.append("/Library/Python/2.7/site-packages/")
    import yaml

or 

    import sys
    sys.append("/Library/Python/2.7/site-packages/PyYAML-3.11-py2.7-macosx-10.10-intel.egg")
    import yaml

prussiap, 2015-01-14:
Not sure which version of BOLTS but it is the latest as of 4 days ago. (not cutting edge but master).

the command is actually
```
import sys
sys.path.append("/Library/Python/2.7/site-packages/PyYAML-3.11-py2.7-macosx-10.10-intel.egg")
import yaml
```

that worked for me.

Now i'm getting a different error (because i didn't install this module I think
```
Traceback (most recent call last):
  File "<input>", line 1, in <module>
  File "/Users/pru**/Library/Preferences/FreeCAD/Mod/BOLTS2/BOLTS/__init__.py", line 40, in <module>
    from gui.freecad_bolts import BoltsWidget
  File "/Users/pru**/Library/Preferences/FreeCAD/Mod/BOLTS2/BOLTS/gui/freecad_bolts.py", line 29, in <module>
    Ui_BoltsWidget,QBoltsWidget = uic.loadUiType(join(bolts_path,'bolts_widget.ui'))
  File "<string>", line 1, in <module>
ImportError: No module named pysideuic
```

peterl94, 2015-01-15:
It is not a symlink. It is a full copy of python, which is necessary to have a self contained application bundle. You can either put the packages there manually, or if you have pip installed you can use **--install-option** to install packages into the bundle.
```
pip install --install-option="--prefix=/Applications/FreeCAD.app/Contents" <package name>
```

jreinhardt, 2015-01-15:
Thanks a lot!

I will link to your post with additional information about where to put packages in the .app, and what to do with a homebrew installation, just for reference.

http://forum.freecadweb.org/viewtopic.php?f=10&t=9322&p=75703#p75713

prussiap, 2015-01-16:
Hi,
Thanks for the feedback Peter and Jreinhardt. So I'm feeling dense but this doesn't seem right. I wanted to copy PyYAML-3.11-py2.7-macosx-10.10-intel.egg to:
```
/Applications/FreeCAD.app/Contents/lib/python2.7/site-packages
```

That went well but then re-opening FreeCad and running import BOLTS still chokes on 
```
Traceback (most recent call last):
  File "<input>", line 1, in <module>
  File "/Users/pru**/Library/Preferences/FreeCAD/Mod/BOLTS2/BOLTS/__init__.py", line 21, in <module>
    import bolttools.blt
  File "/Users/pru**/Library/Preferences/FreeCAD/Mod/BOLTS2/BOLTS/bolttools/blt.py", line 18, in <module>
    import yaml
ImportError: No module named yaml
```

Also i'm having an awful time getting pyside working (I think that's what the pysideuic library is from. I tried to do manual makes, brew, and went through the site.

```
brew install pyside
brew install pyside-tools
```
installed correctly but i can't seem to import the pysideuic or use it.

pointers for both making sure i copied correctly the .egg and the other library.. 

also FYI my FreeCad site-packages folder contains:
```
PySide						numpy						setuptools-0.9.7-py2.7.egg
PyYAML-3.11-py2.7-macosx-10.10-intel.egg	numpy-1.8.1-py2.7.egg-info			setuptools.pth
backports.ssl_match_hostname-3.4.0.2-py2.7.egg	pip-1.4-py2.7.egg				shiboken.so
easy-install.pth				pyparsing-2.0.1-py2.7.egg			six-1.5.2-py2.7.egg
matplotlib-1.3.1-py2.7-macosx-10.6-x86_64.egg	python_dateutil-2.2-py2.7.egg			tornado-3.2-py2.7-macosx-10.6-x86_64.egg
```

sharpcoding, 2018-06-09:
pip utility didn't work for me:
```
pip install --prefix="/Applications/FreeCAD.app/Contents/lib/python2.7" pyyaml
Requirement already satisfied: pyyaml in /Library/Python/2.7/site-packages
```
since there is a bug in the homebrew pip (?)  

The workaround that worked for me: just manually copy the ```/Library/Python/2.7/site-packages/yaml``` folder to ```/Applications/FreeCAD.app/Contents/lib/python2.7/site-packages```

berndhahnebach, 2020-10-20:
since Python 2 is obsolet I will close this issue

closed 2020-10-20 by berndhahnebach