boltsparts/BOLTS_archive #156: No module named pysideuic: FreeCAD 0.14, BOLTS 0.3, Fedora 21

begleysm, 2015-11-08:
Hello,

I'm running FreeCAD 0.14 on Fedora 21 and am attempting to use FreeCAD BOLTS 0.3.

I've put the BOLTS directory and start_bolts.FCMacro in the FreeCAD Macro directory and tried to run the Macro.  This fails with an error in the status bar that I can't entirely read.  So I went to the Python Console and tried to run the macro from there.  It ultimately failed when attemting to import pysideuic.  I then ran "import pysideuic" alone which also failed.  I tried to install python-pyside from the terminal only to find out that it is already installed.  See screen captures for details.

[image]

[image]

Can anyone provide assistance?

Thanks,
begleysm

jreinhardt, 2015-11-08:
I don't know fedora, but on Debian/Ubuntu there is a separate package called pyside-tools, which contains the ui compiler., which needs to be installed as well.

begleysm, 2015-11-08:
Bam, you win the jackpot!

Thanks for the help.  It would be good if that info was included on the Documentation page.
http://www.bolts-library.org/en/docs/0.3/freecad/installation.html

I'm going to modify that webpage and make a pull request when I'm done.

Thanks,
begleysm

jreinhardt, 2015-11-08:
I am glad that worked. Looking forward to the Pull Request.

closed 2015-11-08 by jreinhardt