boltsparts/BOLTS_archive #174: Syntax error in version file of OpenSCAD 0.3 (stable) zip download

Brandano, 2017-09-02:
The licence string in the first line of bolts/common/version.scad is missing string delimiters, which causes OpenSCAD to throw a syntax error when attempting to load the library

gromain, 2019-06-05:
This issue still exists for me.

Illusion65, 2019-09-04:
Thanks! That fixed this issue for me - just put quotes (double or single) around the third parameter. Should be:

function BOLTS_version() = [0, 3, "GPL 3.0"];

RichardWMoran, 2020-02-06:
With reference to function BOLTS_version() = [0, 3, "GPL 3.0"];
Had exactly the same problem from within windows 10 environment, and this fix worked for me, The thing i dont understand is why isnt this corrected within the main Git download, as openscad point to this Library directly, but yet it wont work without this fix for all windows 10 users, so surely someone would have informed this user about this error before now, new to github sorry for any protocol breaches in advance.

berndhahnebach, 2020-10-19:
duplicate of https://github.com/boltsparts/BOLTS/issues/207

closed 2020-10-19 by berndhahnebach