boltsparts/BOLTS_archive #175: SIGSEGV 'segmentation violation' with FreeCAD

thosoo, 2017-09-15:
Using Fedora26 and FreeCAD 0.16, both the bolts 0.3 and dev version evoke a segmentation violation and crash FreeCAD.

FreeCAD log:
`﻿Msg: FreeCAD 0.16, Libs: 0.16RUnknown
© Juergen Riegel, Werner Mayer, Yorik van Havre 2001-2015
  #####                 ####  ###   ####  
  #                    #      # #   #   # 
  #     ##  #### ####  #     #   #  #   # 
  ####  # # #  # #  #  #     #####  #   # 
  #     #   #### ####  #    #     # #   # 
  #     #   #    #     #    #     # #   #  ##  ##  ##
  #     #   #### ####   ### #     # ####   ##  ##  ##

Log: Time = Fri Sep 15 10:19:14 2017
Log: AppDataSkipVendor = true
Log: AppHomePath = /usr/lib64/freecad/
Log: AppIcon = freecad
Log: AppTempPath = /tmp/
Log: BinPath = /usr/lib64/freecad/bin/
Log: BuildRepositoryURL = Unknown
Log: BuildRevision = Unknown
Log: BuildRevisionDate = Unknown
Log: BuildVersionMajor = 0
Log: BuildVersionMinor = 16
Log: CopyrightInfo = © Juergen Riegel, Werner Mayer, Yorik van Havre 2001-2015
  #####                 ####  ###   ####  
  #                    #      # #   #   # 
  #     ##  #### ####  #     #   #  #   # 
  ####  # # #  # #  #  #     #####  #   # 
  #     #   #### ####  #    #     # #   # 
  #     #   #    #     #    #     # #   #  ##  ##  ##
  #     #   #### ####   ### #     # ####   ##  ##  ##

Log: Debug = 0
Log: DocPath = /usr/lib64/freecad/doc/
Log: ExeName = FreeCAD
Log: ExeVendor = FreeCAD
Log: ExeVersion = 0.16
Log: LoggingFile = 1
Log: LoggingFileName = /home/thaison/.FreeCAD/FreeCAD.log
Log: MaintainerUrl = http://www.freecadweb.org/wiki/index.php?title=Main_Page
Log: PATH = /usr/local/bin:/usr/local/sbin:/usr/bin:/usr/sbin:/home/thaison/.local/bin:/home/thaison/bin
Log: PYTHONPATH = 
Log: PythonSearchPath = :/usr/bin/../lib/python27.zip:/usr/bin/../lib64/python2.7:/usr/bin/../lib64/python2.7/plat-linux2:/usr/bin/../lib64/python2.7/lib-tk:/usr/bin/../lib64/python2.7/lib-old:/usr/bin/../lib64/python2.7/lib-dynload
Log: RunMode = Gui
Log: SplashAlignment = Bottom|Left
Log: SplashInfoColor = #c8c8c8
Log: SplashScreen = freecadsplash
Log: SplashTextColor = #ffffff
Log: StartWorkbench = StartWorkbench
Log: SystemParameter = /home/thaison/.FreeCAD/system.cfg
Log: UserAppData = /home/thaison/.FreeCAD/
Log: UserHomePath = /home/thaison
Log: UserParameter = /home/thaison/.FreeCAD/user.cfg
Log: Verbose = 
Log: Create Application
Log: Run App init script
Log: Init: starting App::FreeCADInit.py
Log: Init:   Searching for modules...
Log: Init:      Initializing /usr/lib64/freecad/Mod/Sketcher... done
Log: Init:      Initializing /usr/lib64/freecad/Mod/ReverseEngineering... done
Log: Init:      Initializing /usr/lib64/freecad/Mod/Image... done
Log: Init:      Initializing /usr/lib64/freecad/Mod/PartDesign... done
Log: Init:      Initializing /usr/lib64/freecad/Mod/Ship(Init.py not found)... ignore
Log: Init:      Initializing /usr/lib64/freecad/Mod/Web... done
Log: Init:      Initializing /usr/lib64/freecad/Mod/MeshPart... done
Log: Init:      Initializing /usr/lib64/freecad/Mod/OpenSCAD... done
Log: Init:      Initializing /usr/lib64/freecad/Mod/Fem... done
Log: Init:      Initializing /usr/lib64/freecad/Mod/Spreadsheet... done
Log: Init:      Initializing /usr/lib64/freecad/Mod/Start... done
Log: Init:      Initializing /usr/lib64/freecad/Mod/Draft... done
Log: Init:      Initializing /usr/lib64/freecad/Mod/Test... done
Log: Init:      Initializing /usr/lib64/freecad/Mod/Import... done
Log: Init:      Initializing /usr/lib64/freecad/Mod/Complete... done
Log: Init:      Initializing /usr/lib64/freecad/Mod/Plot(Init.py not found)... ignore
Log: Init:      Initializing /usr/lib64/freecad/Mod/Material... done
Log: Init:      Initializing /usr/lib64/freecad/Mod/Raytracing... done
Log: Init:      Initializing /usr/lib64/freecad/Mod/Mesh... done
Log: Init:      Initializing /usr/lib64/freecad/Mod/Path... done
Log: Init:      Initializing /usr/lib64/freecad/Mod/Part... done
Log: Init:      Initializing /usr/lib64/freecad/Mod/Arch... done
Log: Init:      Initializing /usr/lib64/freecad/Mod/Robot... done
Log: Init:      Initializing /usr/lib64/freecad/Mod/Points... done
Log: Init:      Initializing /usr/lib64/freecad/Mod/Idf... done
Log: Init:      Initializing /usr/lib64/freecad/Mod/Inspection... done
Log: Init:      Initializing /usr/lib64/freecad/Mod/Drawing... done
Log: Using /usr/lib64/freecad/Mod as module path!
Log: System path after init:
Log:    /usr/lib64/freecad/bin
Log:    /usr/lib64/freecad/Mod/Sketcher
Log:    /usr/lib64/freecad/Mod/ReverseEngineering
Log:    /usr/lib64/freecad/Mod/Image
Log:    /usr/lib64/freecad/Mod/PartDesign
Log:    /usr/lib64/freecad/Mod/Ship
Log:    /usr/lib64/freecad/Mod/Web
Log:    /usr/lib64/freecad/Mod/MeshPart
Log:    /usr/lib64/freecad/Mod/OpenSCAD
Log:    /usr/lib64/freecad/Mod/Fem
Log:    /usr/lib64/freecad/Mod/Spreadsheet
Log:    /usr/lib64/freecad/Mod/Start
Log:    /usr/lib64/freecad/Mod/Draft
Log:    /usr/lib64/freecad/Mod/Test
Log:    /usr/lib64/freecad/Mod/Import
Log:    /usr/lib64/freecad/Mod/Complete
Log:    /usr/lib64/freecad/Mod/Plot
Log:    /usr/lib64/freecad/Mod/Material
Log:    /usr/lib64/freecad/Mod/Raytracing
Log:    /usr/lib64/freecad/Mod/Mesh
Log:    /usr/lib64/freecad/Mod/Path
Log:    /usr/lib64/freecad/Mod/Part
Log:    /usr/lib64/freecad/Mod/Arch
Log:    /usr/lib64/freecad/Mod/Robot
Log:    /usr/lib64/freecad/Mod/Points
Log:    /usr/lib64/freecad/Mod/Idf
Log:    /usr/lib64/freecad/Mod/Inspection
Log:    /usr/lib64/freecad/Mod/Drawing
Log:    /usr/local/bin
Log:    /usr/local/sbin
Log:    /usr/bin
Log:    /usr/sbin
Log:    /home/thaison/.local/bin
Log:    /home/thaison/bin
Log: Init: App::FreeCADInit.py done
Log: Init: Creating Gui::Application and QApplication
Log: Local server 'FreeCAD' started
Log: OpenGL version 3.0 or higher is present
Log: Run Gui init script
Log: Init: Running FreeCADGuiInit.py start script...
Log: Init:   Searching modules...
Log: Init:      Initializing /usr/lib64/freecad/Mod/Sketcher... done
Log: Init:      Initializing /usr/lib64/freecad/Mod/ReverseEngineering... done
Log: Init:      Initializing /usr/lib64/freecad/Mod/Image... done
Log: Init:      Initializing /usr/lib64/freecad/Mod/PartDesign... done
Log: Init:      Initializing /usr/lib64/freecad/Mod/Ship... done
Log: Init:      Initializing /usr/lib64/freecad/Mod/Web... done
Log: Init:      Initializing /usr/lib64/freecad/Mod/MeshPart... done
Log: Init:      Initializing /usr/lib64/freecad/Mod/OpenSCAD... done
Log: Init:      Initializing /usr/lib64/freecad/Mod/Fem... done
Log: Init:      Initializing /usr/lib64/freecad/Mod/Spreadsheet... done
Log: Init:      Initializing /usr/lib64/freecad/Mod/Start... done
Log: Init:      Initializing /usr/lib64/freecad/Mod/Draft... done
Log: Init:      Initializing /usr/lib64/freecad/Mod/Test... done
Log: Init:      Initializing /usr/lib64/freecad/Mod/Import... done
Log: Init:      Initializing /usr/lib64/freecad/Mod/Complete... done
Log: Init:      Initializing /usr/lib64/freecad/Mod/Plot... done
Log: Init:      Initializing /usr/lib64/freecad/Mod/Material... done
Log: Init:      Initializing /usr/lib64/freecad/Mod/Raytracing... done
Log: Init:      Initializing /usr/lib64/freecad/Mod/Mesh... done
Log: Init:      Initializing /usr/lib64/freecad/Mod/Path... done
Log: Init:      Initializing /usr/lib64/freecad/Mod/Part... done
Log: Init:      Initializing /usr/lib64/freecad/Mod/Arch... done
Log: Init:      Initializing /usr/lib64/freecad/Mod/Robot... done
Log: Init:      Initializing /usr/lib64/freecad/Mod/Points... done
Log: Init:      Initializing /usr/lib64/freecad/Mod/Idf(InitGui.py not found)... ignore
Log: Init:      Initializing /usr/lib64/freecad/Mod/Inspection... done
Log: Init:      Initializing /usr/lib64/freecad/Mod/Drawing... done
Log: Init: Loading FreeCAD GUI
Log: Init: Running FreeCADGuiInit.py start script... done
Log: Init: Activating default workbench StartWorkbench
Log: Loading GUI of Web module... done
Log: Loading GUI of Start module... done
Log: Loading Start module... done
Log: loaded the Generic plugin 
Log: Init: Showing main window
Log: Main window restored
Log: Show main window
Log: Toolbars restored
Log: Couldn't connect to spacenav daemon
Log: Init: Entering event loop
Log: Init: Processing command line files
Log: Module: Part
Log: Loading Part module... done
Err: *** Abort *** an exception was raised, but no catch was found.Err: 
Err: 	... The exception is:Err: SIGSEGV 'segmentation violation' detected. Address 799Err: `

Benjamin3992, 2017-09-20:
On Arch Linux x64 I get the same error. What hardware are you using? And how can I get such verbose logs?
Python version: Python 2.7.13
AMD FX-8320E
GTX 970

thosoo, 2017-09-20:
I am using an OEM Asus desktop with i3 and HD5670

trillerpfeife, 2018-04-01:
Get the same Error on Arch

berndhahnebach, 2019-06-11:
Would you guys test with latest BOLT installed by FreeCAD addon manager?