boltsparts/BOLTS_archive #176: Website is down

hellerbarde, 2017-10-16:
When I tried to look up the website, i found an invalid certificate and no website.

jreinhardt, 2017-10-17:
That is related to OpenShift shutting down version 2 of their service. I will try to migrate the Website to Openshift V3.

hellerbarde, 2017-10-17:
Let me know if i can help. I just found your project and its fantastic :D

berndhahnebach, 2018-01-16:
hi johannes,

any progress on this?

cheers bernd

BTW BOLTS is still alive in FreeCAD!

kalsan, 2018-04-16:
Well with my current version, the macro start_bolts.FCMacro makes FC crash. I hope that either (or even better both) website and plugin will work again soon

luzpaz, 2018-09-09:
@jreinhardt I'd like to help keep the website going. Can you explain how I'd need to to do that?

berndhahnebach, 2019-05-09:
any updates ... ?

some links:
https://github.com/berndhahnebach/BOLTSFC/issues/1
https://github.com/jreinhardt/BOLTS/issues/176
https://forum.freecadweb.org/viewtopic.php?f=8&t=30668

berndhahnebach, 2019-05-16:
good news BOLTS and BOLTSFC have a new home on the organization https://github.com/boltsparts/ ATM the founder of BOLTS and myself are member if the organization.

The website will be next, but this will for sure be a bigger chunk.

jreinhardt, 2019-05-29:
The website is up again here: https://boltsparts.github.io/

I will look into getting the boltslibrary.org domain name set up again.

closed 2019-06-06 by berndhahnebach