boltsparts/BOLTS_archive #177: Missing instructions to generate the website locally

galou, 2019-02-06:
The website being down people (including me) willing to generate the website locally miss instructions to do so because the documentation link is on the website itself. Please anyone who knows how to do it put the instruction as comments.

jreinhardt, 2019-02-06:
In the root folder there is a script called bolts.py, with which tables can be exported to html.

The website itself is written in Flask: http://flask.pocoo.org/, the instructions there should more or less work from the website directory.

galou, 2019-02-07:
Here is what I tried without success:

with `python2 bolts.py export html`:
```
Traceback (most recent call last):
  File "bolts.py", line 347, in <module>
    args.func(args)
  File "bolts.py", line 59, in export
    from backends.html import HTMLBackend
ImportError: No module named html
```
Also, from within the website directory `FLASK_APP=html.py flask run`:
```
 * Serving Flask app "html.py"
 * Environment: production
   WARNING: Do not use the development server in a production environment.
   Use a production WSGI server instead.
 * Debug mode: off
Usage: flask run [OPTIONS]

Error: While importing "website.html", an ImportError was raised:

Traceback (most recent call last):
  File "/home/gael/.local/lib/python2.7/site-packages/flask/cli.py", line 235, in locate_app
    __import__(module_name)
  File "/home/gael/.cache/cesnet-owncloud/04-sources/freecad/bolts/website/__init__.py", line 2, in <module>
    from flask.ext.babelex import Babel,format_datetime, gettext, lazy_gettext, Domain
ImportError: No module named ext.babelex
```

I installed `python-flask` version 1.0.2 and `Flask-BabelEx` version 0.9.3 from `pip` on Ubuntu 18.04.

I have to admit that I'm lost. I'll be thankful for any hint!

galou, 2019-02-07:
I made some progresses by replacing all occurrences of type `babel.ext.SOMETHING` with `babel_SOMETHING` and have now a new error:
```
KeyError: 'OPENSHIFT_REPO_DIR'
```
because `OPENSHIFT_REPO_DIR` is not defined in my environment. To what should I set it?

galou, 2019-03-21:
Would it be possible to have a description of the steps needed to generate the web pages and, possibly, serve them locally?

Thanks a lot!

jreinhardt, 2019-05-26:
Hi,

I finally started to look into the website and decided to turn it from a Flask app into a static site generator (not such a big step thanks to https://pythonhosted.org/Frozen-Flask/).

If you want you can try it: Check out the website rebuild branch (https://github.com/boltsparts/BOLTS/tree/website-rebuild)

and in the repo root something like

virtualenv venv
. venv/bin/activate
pip install -r requirements
python bolts.py test website

should serve the website on http://127.0.0.1:5000/

python bolts.py export website

should generate static html and puts it into output/website

galou, 2019-05-27:
Thanks a lot! It seems to work well.

berndhahnebach, 2021-09-10:
the webpage works, but the information presented here should be documented somewhere. thus I do not close it.

berndhahnebach, 2022-11-01:
to be continued here ... https://github.com/boltsparts/BOLTS/issues/258

closed 2022-11-01 by berndhahnebach

berndhahnebach, 2022-11-01:
related https://github.com/boltsparts/boltsparts.github.io/issues/3