boltsparts/BOLTS_archive #189: v0.4

berndhahnebach, 2019-06-11:
Johannes,

we should release a 0.4 ASAP. On the website the documentation for installation points to download on github and there we have 0.3 to download.

bernd

berndhahnebach, 2019-06-11:
What is needed actually:

- set the tag commit 
- upload packages to github
- make a blogpost
- generate BOLTSFC

anything else?

berndhahnebach, 2021-09-10:
Webpage:
- make 0.4 the official release, Where in the webpage?

Code:
- after the  release of 0.5 change the version number to the next version ( 0.9 ;-) )
- Where in the code, hopefully only on one string ...

luzpaz, 2022-01-25:
@berndhahnebach any progress on this?

berndhahnebach, 2022-10-31:
0.4.1 is even released already .... https://github.com/boltsparts/BOLTS/releases

I leave it open since it documentes the release process. I will close it if it is documented somewhere else.