boltsparts/BOLTS_archive #207: Parser error in file "common/version.scad"

alnavv, 2020-02-06:
[ ERROR ]
ERROR: Parser error in file "/tmp/common/version.scad", line 1: syntax error 
ERROR: Compilation failed

[ WORKING ENVIRONMENT ]
openscad in docker, and all the libraries inside the docker image (/LIBS/BOLTS, )
From the openscad Help/Library Info:
OPENSCADPATH: /LIBS:/LIBS/BOLTS
OpenSCAD library path:
/LIBS
/LIBS/BOLTS

in my host, i have the file:
/tmp/test.scad
with "ONLY": 
include <BOLTS.scad>
so i open this file using the container and get the described error.

NOTE: All the other libraries work fine (NopScadLib, dotSCAD, BOSL, etc ..)

Brandano, 2020-02-10:
Pretty sure this is STILL the same issue I opened three years ago:
https://github.com/boltsparts/BOLTS/issues/174
Just add string delimiters around the licence description string.

alnavv, 2020-02-12:
Thanks, Brandano, it seems you are right.  Just opened the file inside the container and the error was there ... addedd delimiters and the error was gone.
It would be nice to fix this :)

berndhahnebach, 2020-10-19:
I do not use openscad at all. Would one of you go for a PR and

berndhahnebach, 2020-10-19:
this might have been fixed already ... https://github.com/boltsparts/BOLTS/blob/da747e25ae66a78d6f1f7b0a1d926ebd3ba12e16/backends/openscad.py#L201 IMHO quotes should be added on openscad BOLTS.

What did you use? Did you download version 0.3 for openscad or did you created the version from BOLTS source code yourself?

bernd

berndhahnebach, 2020-10-19:
It actually is fixed. https://github.com/boltsparts/BOLTS/commit/5f6592b8e1e62e0453d3d3ed71be112fa4bcfe75

I may create a uptodate version for openscad but I can not test it, since I do not use it.

berndhahnebach, 2020-10-20:
would you guys test: https://github.com/boltsparts/BOLTS/releases/tag/dd53a1d

Brandano, 2020-10-20:
For the purpose of this bug, it seems to work fine now. I do get an unrelated error in base/hex_socket/hex_socket.scad:
WARNING: Ignoring unknown variable 'h', in file libs/bolts/base/hex_socket/hex_socket.scad, line 48. 
Which means that this check is skipped: BOLTS_check_dimension_positive(h,"l too short");
Probably hex_socket_2 is missing one line like line 24 in the module immediately preceding it?

Brandano, 2020-10-20:
Or maybe the check should be on k, since it's used as h on line 53

berndhahnebach, 2020-10-20:
I have created a new issue in this regard ... https://github.com/boltsparts/BOLTS/issues/220

will close this one.

closed 2020-10-20 by berndhahnebach