boltsparts/BOLTS_archive #211: No module named pysideuic

alfredopalhares, 2020-04-22:
Hello,

I installed the BOLTS Library on FreeCAD and restarted as It shown the documentation. The macro did not execute, so I have tried to import the module on the console:
```
>>> import BOLTS
Traceback (most recent call last):
  File "<input>", line 1, in <module>
  File "/home/masterkorp/.FreeCAD/Mod/BOLTSFC/BOLTS/__init__.py", line 44, in <module>
    from .gui import freecad_bolts as boltsgui
  File "/home/masterkorp/.FreeCAD/Mod/BOLTSFC/BOLTS/gui/freecad_bolts.py", line 30, in <module>
    Ui_BoltsWidget,QBoltsWidget = uic.loadUiType(join(bolts_path,'bolts_widget.ui'))
  File "<string>", line 1, in <module>
ModuleNotFoundError: No module named 'pysideuic'
````
According to the documentation, i need to have the pyside tools package installed, which I do.
```
-> % pacman -Qs pyside
local/pyside2 5.14.2-1 (qt qt5)
    Enables the use of Qt5 APIs in Python applications
local/pyside2-tools 5.14.2-1
    Tools for PySide2
local/python2-pyside2 5.12.1-1 (qt qt5)
    Enables the use of Qt5 APIs in Python applications
```

Yet I cannot indeed use that module:
```
-> % python3
Python 3.8.2 (default, Apr  8 2020, 14:31:25)
[GCC 9.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import PySide2.QtCore
>>> import PySideUic
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ModuleNotFoundError: No module named 'PySideUic'
>>> import pysiduic
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ModuleNotFoundError: No module named 'pysiduic'
>>>
```
Any ideas?

blardyblar, 2020-05-28:
I believe this is because you've installed pyside2-tools and not pyside-tools. Not sure what o/s you're using, but it's the same in Debian SID. It seems like they want pyside-tools which hasn't been in debian since Strech, which is extremely old.

The problem is it comes from inside FreeCADGui module, not from bolts directly:

`        from FreeCADGui import PySideUic as uic

        try:
                Ui_BoltsWidget,QBoltsWidget = uic.loadUiType(join(bolts_path,'bolts_widget.ui'))
                Ui_ValueWidget,QValueWidget = uic.loadUiType(join(bolts_path,'value_widget.ui'))
                Ui_BoolWidget,QBoolWidget = uic.loadUiType(join(bolts_path,'bool_widget.ui'))
                Ui_TableIndexWidget,QTableIndexWidget = uic.loadUiType(join(bolts_path,'tableindex_widget.ui'))
                Ui_PropertyWidget,QPropertyWidget = uic.loadUiType(join(bolts_path,'property_widget.ui'))
        except ImportError:
                FreeCAD.Console.PrintError("uic import failed. Make sure that the pyside tools are installed")
                raise`

berndhahnebach, 2020-10-19:
https://github.com/boltsparts/BOLTSFC/issues/26

berndhahnebach, 2020-10-19:
> 
> be very careful about the tab indenting in this file - my vim was set to insert spaces for tabs - completely messed up the indenting and python wouldn't run the file until I fixed it
>

something that anoys me all the time the tabs in BOLTS code. On many places I did exchanged them already, but one has to be careful with this to not break something.

bernd

berndhahnebach, 2020-10-19:
Since it has nothing to do with BOLTS I will close it. 

BTW:
thanks for posting the workaround

closed 2020-10-19 by berndhahnebach