boltsparts/BOLTS_archive #214: Unknown variable center warning in aluminiumProfiles.scad

olemis, 2020-06-16:
**BOLTS library version**: 0.3
**OS** : Darwin Kernel Version 17.4.0: Sun Dec 17 09:19:54 PST 2017; root:xnu-4570.41.2~1/RELEASE_X86_64

**Code snippet**

```
include <BOLTS.scad>
tslot20x20_2s(detailed=true);
```

**Result**

```
WARNING: Ignoring unknown variable 'center', in file $OPENSCADPATH/libraries/base/aluminiumProfiles.scad, line 269.
```

**Expected behavior**

No warning

**Additional context**

The issue is caused by a ternary condition evaluated and no way to resolve `center` variable name to a value in local module namespace.

berndhahnebach, 2020-10-19:
Would you fix it and go for a PR?  I do not use OpenScad at all.  Thus fixing any bug there is quite difficault for me.

berndhahnebach, 2020-10-20:
would you test: https://github.com/boltsparts/BOLTS/releases/tag/dd53a1d ?