boltsparts/BOLTS_archive #224: Fails to load with Freecad ~0.19

wavexx, 2020-12-06:
I've just tried to start BOLTSFC on the latest experimental build of Freecad 0.19 in debian. Installed using the addon manager, but it fails:

```
17:58:55  uic import failed. Make sure that the pyside tools are installed
Traceback (most recent call last):
  File "~/.FreeCAD/Macro/start_bolts.FCMacro", line 1, in <module>
    import BOLTS
  File "~/.FreeCAD/Mod/BOLTSFC/BOLTS/__init__.py", line 44, in <module>
    from .gui import freecad_bolts as boltsgui
  File "~/.FreeCAD/Mod/BOLTSFC/BOLTS/gui/freecad_bolts.py", line 37, in <module>
    Ui_BoltsWidget, QBoltsWidget = uic.loadUiType(
  File "<string>", line 1, in <module>
<class 'ModuleNotFoundError'>: No module named 'pyside2uic'
```

I'm also using several other addons that make use of FreeCADGui.PySideUic (one example is "Reinforcement"), but it works correctly. Not sure what is going on..

Davide-sd, 2021-03-20:
Having the same problem. Did you find a workaround?

wavexx, 2021-03-21:
I've been using "Fasteners" instead :/

s-light, 2021-06-13:
i think this is related:
https://forum.freecadweb.org/viewtopic.php?p=440973#p440973
use the binaries that have moved to qt5 internal.. ??!