boltsparts/BOLTS_archive #225: common\version.scad   has an error when loading

toothrobber, 2021-01-25:
I just downloaded the the BOLTS library for OpenScad (OpenSCAD version 2019.05 
Copyright (C) 2009-2019 The OpenSCAD Developers)

I placed the BOLTS library in Resources\libraries\BOLTS

This sample code failed. with an error in the version.scad file
include <BOLTS/bolts.scad>
ISO7089("M6");

GPL 3.0 was not in quotes, this caused an error 

The corrected line is as follows.

function BOLTS_version() = [0, 3, "GPL 3.0"];

aleksey-ru52, 2021-01-29:
confirm
just downloaded version BOLTS_OpenSCAD_0.3_gpl3.tar.gz
it has the same error
@toothrobber thanks in advice

berndhahnebach, 2021-08-30:
in openscad version created with https://github.com/boltsparts/BOLTS/commit/72676c7d413e444fb01bf8279a2b3337c560f894 the version.scad file looks like this ...

```
function BOLTS_version() = "development";
function BOLTS_date() = [2021,8,30];
function BOLTS_license() = "LGPL 2.1";

```

if this is fine, I would close this issue ...

berndhahnebach, 2022-10-24:
could anyone of openscad guys confirm if this has been fixed?