boltsparts/BOLTS_archive #240: add PythonPackage to pypi

berndhahnebach, 2021-09-10:


berndhahnebach, 2021-09-10:
bolts for Python would be intallable just by

pip install boltspy

berndhahnebach, 2021-09-10:
may be even add a repository for the files needed for pypi

they are on my local machine only ATM

berndhahnebach, 2021-09-10:
https://pypi.org/project/boltspy/

Andrej730, 2022-10-19:
Bumping suggestion - it's important to add both `data` and `pythonpackage` to python package since otherwise boltspy is raising exceptions if you just install it with `pip install boltspy` and then try to `import boltspy`.

Or atleast mention on PyPi page those folders need to be copied from repository to package's folder (e.g. `%localappdata%\Programs\Python\Python310\Lib\site-packages\boltspy\data`) to avoid errors.