boltsparts/BOLTS_archive #248: Website is dead.

GAZ082, 2022-05-14:


wamcvey, 2022-10-08:
I was frustrated by all the docs being inaccessible and then noticed that the @boltparts org has another repo for their website which mentioned: https://boltsparts.github.io/en/index.html  which seems to be where the current documentation has moved to.

berndhahnebach, 2022-10-24:
The link you posted it the github pages link. Actually you can use the short version too ... https://boltsparts.github.io/ All other should show the contents of this github page, even if in the adress list still is the other one. Somehow all the redirects do not work anymore ...

berndhahnebach, 2022-10-24:
The problem is http://www.bolts-library.org/ is down for some reason. The domain is owned by @jreinhardt. He is the only one who knows what the problem is.

The following domain would work too, but is just a redirect to the github page ... http://www.boltsparts.org This one is owned by me.