boltsparts/BOLTS_archive #251: Addition of Australian profiles and IFC backend

Moult, 2022-10-17:
Hey @berndhahnebach (or other maintainer?), I'm looking to sponsor some work by @Andrej730 to integrate BOLTS into the BlenderBIM Add-on, and also in the process to add Australian steel profiles. I assume this involves editing the .blt collections in https://github.com/boltsparts/BOLTS/tree/master/data . The parts will include hot rolled I, C, L, RHS, CHS, SHS, and cold Cees and Zeds. Yaml files will be added for all the dimensions.

What about:

 - Where are structural area properties stored? Or are there calculation functions available?
 - Is there some existing IFC code written? Perhaps by @Jesusbill ?
 - Is there any docs on boltspy?

Moult, 2022-10-19:
Or I guess ping @jreinhardt ? I'm not quite sure who the maintainer is (would be happy to help out, if this project is feeling lonely? :)

berndhahnebach, 2022-10-24:
https://github.com/boltsparts/BOLTS/pull/252

I will have a look.

berndhahnebach, 2022-10-24:
australian are merged

berndhahnebach, 2022-10-31:
Could we close this?

closed 2022-11-02 by berndhahnebach