boltsparts/BOLTS_archive #257: developer meeting Oktober 2022

berndhahnebach, 2022-10-31:
### Member
+ @berndhahnebach 
+ @Moult
+ @Jesusbill

### decisions
#### Bernd
+ updates boltsfc and fixes standard names and standard body of Australien profiles to fit in Standard nameing system
+ tries to create BOLTS website and postes problem, tries to solve https://github.com/boltsparts/BOLTS/issues/185
+ solves https://github.com/boltsparts/BOLTS/issues/163
+ gets in contact with @jreinhardt in the regard of https://github.com/boltsparts/BOLTS/issues/248
    + done but no aswer so far

### Dion Moult
+ add Austrailian "standard" furniture spaces
+ tries to get these for South Africa as well
+ uses Standard naming to fit in the Standard naming
+ add here https://github.com/boltsparts/BOLTS/blob/master/pythonpackage/readme.md some new part as reference to the script of IfcOpenShell to create IFC Type libraries
+ might can trigger some resources if bernd is not able to create the website

### Ioannis
+ adds American profiles https://github.com/boltsparts/BOLTS/issues/243 , uses inch unit which is available but not used ATM.
+ I am curious if it works ... 

### No decision made yet, just discussion
#### MetaData
+ adding MetaData to the tables like Area, Center of Gravitiy and so on
+ It might be smarter to create new tables for Metadata instead of just adding columns
#### BOLTS file structure
+ improve file structure
+ may be move backend data creation, and backend readme directories (they are in first file layer) into backend, because it is backend data

thanks guys, cheers bernd

berndhahnebach, 2022-11-01:
https://github.com/boltsparts/BOLTS/issues/258#issuecomment-1298493805

berndhahnebach, 2022-11-01:
[image]

berndhahnebach, 2022-11-01:
ad501d9453da04e3f345b5d19b4f8663d134909b

Moult, 2022-11-01:
Architectural things WIP: https://docs.google.com/spreadsheets/d/1HVwG0P0aEwq5p7SvRL1Q_IqJURqZJ5T5NF4rj9S5mVY/edit?usp=sharing

BTW a recent requirement by my company is to be able to lookup some mass per volume measurements for Australian profiles to be used in LCA calculations. So I wonder what the pros and cons are of adding extra columns:

Pros of adding columns:

 - Easy
 - Data all in one place

Cons of adding columns:

 - Unclear to user which are required for dimensional (but user can read descriptions / units)
 - Constrains metadata to 1:1, cannot have 1:N relationships of metadata

What's the alternative to adding columns? Groups of metadata as a separate yaml? Sharing the same key as the first column?

Jesusbill, 2022-11-03:
W profiles with imperial units added and seem to work for the backend.
I have used AISC as the standard, not sure if it is in the list, maybe this has to change.

berndhahnebach, 2022-11-03:
There is no list for standard, just write what you would like too.

Jesusbill, 2022-11-04:
ok then I think AISC is ok, although not a standard organization