boltsparts/BOLTS_archive #258: recreating the website does make problems

berndhahnebach, 2022-11-01:
github deployment gives an error ... 

https://github.com/boltsparts/boltsparts.github.io/actions/runs/3365082788/jobs/5580174877

https://github.com/boltsparts/boltsparts.github.io/commit/797f0bc4653ac46bc2b39558ee32bd91a567e73a

berndhahnebach, 2022-11-01:
some hints might be here too ... https://github.com/boltsparts/BOLTS/issues/177

berndhahnebach, 2022-11-01:
do not forget to solve https://github.com/boltsparts/BOLTS/issues/185

berndhahnebach, 2022-11-01:
it works ... https://boltsparts.github.io/en/blog/2022/11/1/Development-Continuation.html
but the process is still not documented ... thus I leave it open

Moult, 2022-11-01:
Hooray! Once it's documented we can check whether or not we can replicate it too.

berndhahnebach, 2022-11-08:
from this post ... https://github.com/boltsparts/BOLTS_archive/pull/184#issuecomment-498116318

- install a local webserver (I have installed lighttpd), no need to config anything
- on my debian I need to use python3 instead of python
- create the website backend
```
python bolts.py export website

```

> To test the website locally you can serve the directory in a local web server by executing
> 
> ```
> python3 -m http.server 8000 --bind 127.0.0.1
> ```
> 
> from the output/website directory. You can then point your browser to 127.0.0.1:8000
> 
> Or you use
> 
> ```
> python3 bolts.py test website
> ```
> 
> which serves the unfrozen flask application on 127.0.0.1:5000.

berndhahnebach, 2022-11-08:
the above works great for me ...