boltsparts/BOLTS_archive #263: automate the process of updating dependencies of BOLTS

berndhahnebach, 2022-11-02:
if a commit was made to BOLTS the repos BOLTSFC and boltsparts.github.io should be updated automatically.

The only idea I have is a Python script which uses git and github python modules. For adding FreeCAD PullRequest branches to the ci i did something in this regard ... https://github.com/FreeCAD/FreeCAD-CI/tree/main/gitlab/scripts

In this case here the script would be manually run by a maintainer after he merges some code to BOLTS

+ export BOLTS freecad backend
+ if there is no error
    + copy backend into local BOLTSFC reop
    + commit
    + push to github
+ export BOLTS website backend
    + copy backend into local website repo
    + commit
    + push to github

the dev has to make sure if everythings work before run the script. This is something he has to do anyway, no matter if the process above is automated or not.

berndhahnebach, 2022-11-02:
for sure something for openscad and pythonpackage could be done as well

berndhahnebach, 2022-11-04:
done for boltsfc and website :-) very cool :-) boltsfc and websit repo are updated in a few seconds :-)

berndhahnebach, 2022-11-04:
BOLTSFC, boltswebsite repo, boltspy and boltsos are automatically updated by a script which I run on any change on boltsparts.

updateing pip still needs to do manually. Is this possible by python to?

Moult, 2022-11-08:
https://github.com/IfcOpenShell/IfcOpenShell/blob/v0.7.0/.github/workflows/ci-ifcopenshell-pypi.yml#L68-L73 you can use github actions to publish to PyPI but I do not recommend doing it on every commit since there is a filesize limit, it's better to do it once a month.

berndhahnebach, 2022-11-08:
wow, very cool. I will have a look at this. Once a month would be fine too.